fivdi/i2c-bus #96: i2c out, can't accept value outside of -128 to +127?

Nodi-Rubrum, 2020-08-12:
i2c out, can't accept value outside of range -128 to +127?  To fully power on a tsl2591 sensor, required to send to command/register 0, a value of 147, which is well with in the 0 to 255 value range for a single byte.  Is the following error a limit of the i2c-bus code?  I receive the following error:
`"RangeError: The value of "value" is out of range. It must be >= -128 and <= 127. Received 147"`

fivdi, 2020-08-12:
I don't think the source of this error is code in the i2c-bus module. Please post the complete stack trace so that it's possible to see where the error was thrown.

Nodi-Rubrum, 2020-08-12:
Will do.  Given I am using the i2c-bus library via a Node Red node that in turn uses i2c-bus library, I have also contacted the author of the Node Red node as well.

fivdi, 2020-08-18:
The reason this error occurs is explained at https://github.com/nielsnl68/node-red-contrib-i2c/issues/26. See also 
https://discourse.nodered.org/t/is-there-a-range-limit-for-node-red-contrib-i2c-i2c-out-node/31458.

Closing in favour of https://github.com/nielsnl68/node-red-contrib-i2c/issues/26.

closed 2020-08-18 by fivdi

Nodi-Rubrum, 2020-08-19:
Yup, I found the above not long after I posted here, moreover, walking your code, I did not find anything that supported the issue in your code, but I wanted to confirm explicitly, not assume anything, in case I missed something.