fivdi/i2c-bus #97: How to list available adapters?

KnorpelSenf, 2020-08-17:
What is the equivalent of this?
```
i2cdetect -l
```
I would like to find out which numbers I can use for the number of the I2C bus/adapter to open with the `open` function. Thanks

fivdi, 2020-08-17:
i2c-bus doesn't have an API for this purpose. However, [glob](https://www.npmjs.com/package/glob) and regular expressions could be used to retrieve a list of I2C bus numbers. Something like this should do it:

```js
const glob = require('glob');

const busNumbers = glob.sync('/dev/i2c-*').
  filter(fileName => fileName.match(/\/i2c-\d+$/) != null).
  map(fileName => parseInt(fileName.match(/\d+$/)[0], 10));

console.log(busNumbers);
```
Here glob is used synchronously but it can also be used asynchronously. `busNumbers` is an array of numbers.

KnorpelSenf, 2020-08-17:
Awesome, thanks for the great code example! I was considering doing something in that direction if the lib would not have it built-in, but it would've needed some time to figure it out.

closed 2020-08-17 by KnorpelSenf

PizzaProgram, 2021-09-04:
Exactly what I was looking for 2 days long.
This should go to the Example dir.

PizzaProgram, 2021-09-04:
Sorry, I was happy too soon :-(
Please help me a bit, because the code above is not working.

1. Installed 'globe': $ npm i glob
2. restarted NodeRed: $ node-red-restart
3. Re-deployed node
... but still the same error:

`"TypeError: Cannot read property 'sync' of undefined"`

And yes, I've changed first line to:

```javascript
const glob = global.get('glob');
```

PizzaProgram, 2021-09-04:
... SOLVED!

1. First this file needs to be edited : /home/pi/.node-red/settings.js 
to add 'glob' as reference:
```javascript
functionGlobalContext: {
  i2c:require('i2c-bus'),
  glob:require('glob')
}
```
2. and of course a:
 $ node-red-restart

fivdi, 2021-09-05:
@PizzaProgram good to hear that it's working now.