fivdi/i2c-bus #98: Error

ELEliseev, 2020-10-20:
Hi!
I build project use electronjs and was be good (version i2c-bus 5.1.0) but
I get fault until build tomorrow  (version i2c-bus 5.1.0 and higher)

> electron .

App threw an error during load
Error: The module '/home/.../node_modules/i2c-bus/build/Release/i2c.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 73. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func (electron/js2c/asar.js:155:31)
    at process.func [as dlopen] (electron/js2c/asar.js:155:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:828:18)
    at Object.func (electron/js2c/asar.js:155:31)
    at Object.func [as .node] (electron/js2c/asar.js:155:31)
    at Module.load (internal/modules/cjs/loader.js:645:32)
    at Function.Module._load (internal/modules/cjs/loader.js:560:12)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at bindings (/home/.../node_modules/bindings/bindings.js:112:48)
A JavaScript error occurred in the main process
Uncaught Exception:
Error: The module '/home/.../node_modules/i2c-bus/build/Release/i2c.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 73. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func (electron/js2c/asar.js:155:31)
    at process.func [as dlopen] (electron/js2c/asar.js:155:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:828:18)
    at Object.func (electron/js2c/asar.js:155:31)
    at Object.func [as .node] (electron/js2c/asar.js:155:31)
    at Module.load (internal/modules/cjs/loader.js:645:32)
    at Function.Module._load (internal/modules/cjs/loader.js:560:12)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at bindings (/home/.../node_modules/bindings/bindings.js:112:48)
The old Assembly folder node-modules/i2c-bus is working well.I tried various versions of nodejs npm and electronjs but all without success.And the old version 5.1.0 stopped being built. when building, I noticed that node-gyp is turned on, I don't remember if it was like this before.I tried it on different computers,

fivdi, 2020-10-23:
I'm afraid I'm not familiar with Electron and don't know if I can provide much help. However, i2c-bus is a native Node module and special steps need to be taken when installing native Node modules. For additional information see [here](https://www.electronjs.org/docs/tutorial/using-native-node-modules).

ELEliseev, 2020-10-23:
Thank you very much

closed 2020-10-23 by ELEliseev

shenzhuxi, 2022-02-27:
https://www.electronjs.org/docs/latest/tutorial/using-native-node-modules here explained why it threw the error. I successfully rebuild i2c-bus with command electron-rebuild. 

However I can only read once or twice from i2c in electron main.js and it will freeze the app.