fivdi/i2c-bus #99: `npm install i2c-bus` fails on macOS Big Sur (11.1)

jorisvervuurt, 2020-12-21:
When trying to install `i2c-bus` (v5.2.1, I get the following error on macOS Big Sur (11.1).
I'm using Node.js 14.15.3 LTS through `nvm`.

```
gyp: Call to 'node -e "require('nan')"' returned exit status 0 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/Users/jorisvervuurt/.nvm/versions/node/v14.15.3/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Darwin 20.2.0
gyp ERR! command "/Users/jorisvervuurt/.nvm/versions/node/v14.15.3/bin/node" "/Users/jorisvervuurt/.nvm/versions/node/v14.15.3/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/jorisvervuurt/Documents/Software/GitHub/SUPER_SECRET/node_modules/i2c-bus
gyp ERR! node -v v14.15.3
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
```

fivdi, 2020-12-21:
Can you post the complete output of the command `npm install i2c-bus` please?

jorisvervuurt, 2020-12-21:
Of course, here's the entire output:

```
jorisvervuurt@Mac-mini-van-Joris-Vervuurt SUPER_SECRET % npm install i2c-bus
Debugger listening on ws://127.0.0.1:50889/91f951b6-5c9a-4a2f-a2a0-bab086e60853
For help, see: https://nodejs.org/en/docs/inspector
Debugger attached.

> i2c-bus@5.2.1 install /Users/jorisvervuurt/Documents/Software/GitHub/SUPER_SECRET/node_modules/i2c-bus
> node-gyp rebuild

Debugger listening on ws://127.0.0.1:50897/b8b76c90-cc7f-4249-b768-2c98271c2e82
For help, see: https://nodejs.org/en/docs/inspector
Debugger attached.
Debugger listening on ws://127.0.0.1:50901/a0c58568-93c9-4b1d-8590-96ea921ceb29
For help, see: https://nodejs.org/en/docs/inspector
Debugger attached.
Waiting for the debugger to disconnect...
gyp: Call to 'node -e "require('nan')"' returned exit status 0 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/Users/jorisvervuurt/.nvm/versions/node/v14.15.3/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Darwin 20.2.0
gyp ERR! command "/Users/jorisvervuurt/.nvm/versions/node/v14.15.3/bin/node" "/Users/jorisvervuurt/.nvm/versions/node/v14.15.3/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/jorisvervuurt/Documents/Software/GitHub/SUPER_SECRET/node_modules/i2c-bus
gyp ERR! node -v v14.15.3
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
Waiting for the debugger to disconnect...
npm WARN SUPER_SECRET@0.0.9 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! i2c-bus@5.2.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c-bus@5.2.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/jorisvervuurt/.npm/_logs/2020-12-21T12_12_55_150Z-debug.log
Waiting for the debugger to disconnect...
```

Let me know if you need the `debug.log` file.

jorisvervuurt, 2020-12-21:
Okay... this is weird. I retried removing and installing the package and now it went fine.
Not sure what was the issue, but it seems to work fine now so I'm closing the issue. ;-)

closed 2020-12-21 by jorisvervuurt

fivdi, 2020-12-21:
Ok, good to hear that it works now. Thanks for the feedback.

SPlatten, 2021-03-09:
@jorisvervuurt , what did you do to fix this, I'm having the same issue, I am using Big Sur 11.2.3, and when I try to install node-gyp:

npm install -g node-gyp

The result is:
npm ERR! code ENOTFOUND
npm ERR! syscall getaddrinfo
npm ERR! errno ENOTFOUND
npm ERR! network request to https://registry.npmjs.org/node-gyp failed, reason: getaddrinfo ENOTFOUND myproxy.example.com
npm ERR! network This is a problem related to network connectivity.
npm ERR! network In most cases you are behind a proxy or have bad network settings.
npm ERR! network 
npm ERR! network If you are behind a proxy, please make sure that the
npm ERR! network 'proxy' config is set properly.  See: 'npm help config'

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/sy/.npm/_logs/2021-03-09T15_25_52_688Z-debug.log

I am not behind a proxy.

jorisvervuurt, 2021-03-09:
@SPlatten the error you are seeing is not related to the error I experienced. It seems either the npm registry was temporarily unreachable or you are experience internet connectivity issues.