fivdi/i2c-bus #101: Support for 16 bit command/index

rgillan, 2021-06-03:
In interfacing with some advanced sensors (e.g. ST VL53L3X) it requires a 16 bit command to identify the register to access.  We were wondering if it was possible to add this functionality to this module (which we use extensively thank you).

[image]
[image]

rgillan, 2021-06-03:
although we have been able to do some basic comms with i2cWriteSync and i2cReadSync, so probably more for convenience than anything I expect.

fivdi, 2021-06-03:
The plain I2C methods described in the readme can be used to achieve what's needed here.

For example, for the read shown in Figure 12, call `i2cWrite` or `i2cWriteSync` to write to address 0x52 followed by a call to `i2cRead` or `i2cReadSync` to read from address 0x53.

For the write shown in Figure 11, call `i2cWrite` or `i2cWriteSync` to write to address 0x52.

If needed, your application can implement higher level methods suitable for accessing the VL53L3X which leverage the functionality provided by the plain I2C methods that i2c-bus offers.

I won't be adding methods to cover this use case to i2c-bus. The read shown in Figure 12 that first writes to address 0x52 and then reads from address 0x53 isn't very conventional.

rgillan, 2021-06-03:
Thanks for getting back so promptly.  So we've worked out the 52/53 thing is actually a documentation anomaly.  It's actually write/read to 0x29 including the r/w bit (silly way to identify the address in my opinion).  Anyway, thanks for this great module and we'll make do with the plain i2c methods.  Cheers

closed 2021-06-03 by rgillan