fivdi/i2c-bus #102: Does this library support different speeds?

eltoroit, 2021-06-22:
Hi,

Thanks for building a great library. I am converting code from RPIO to this library and it's going well, except that I can't seem to figure out how to convert `rpio.i2cSetBaudRate(this.BAUD_RATE);`

The I2C protocol supports four clock speeds
Standard-mode (100 Kbit/s)
Fast-mode (400 Kbit/s)
Fast-mode Plus (1 Mbit/s)
High-speed (3.4 Mbit/s)

But I am not finding out how to change those in your library.

Thanks

fivdi, 2021-06-22:
i2c-bus was implemented to function on as may Linux boards as possible rather than just the Raspberry Pi. For example, it also functions on the BeagleBone Black. Unfortunately, there isn't a standard way to set the I2C baudrate available on Linux. Each system does it differently. On a Raspberry Pi using Raspberry Pi OS, the baudrate can be set with the `i2c_arm_baudrate` parameter in the configuration file `/boot/config.txt`. As an example, to set the I2C baudrate to 400 Kbit/s add the following line to ` /boot/config.txt`

```
dtparam=i2c_arm_baudrate=400000
```

The default baud rate on the Raspberry Pi when using Raspberry Pi OS is 100 Kbit/s.

Reboot the Raspberry Pi after modifying `/boot/config.txt`.

fivdi, 2021-07-01:
I'll go ahead and close this issue as there is no TODO for i2c-bus.

closed 2021-07-01 by fivdi