fivdi/i2c-bus #105: .openSync() works async way

PizzaProgram, 2021-09-14:
Hi,

I'm fixing / enhacing currently an MPC chip node for NodeRed: https://github.com/PizzaProgram/mcp23017chip
I'm inserting lot's of try .. catch lines to make the old code of @willhey stable.

# But it seems, the code runs a few lines more, and does not stop after at .openSync error !

```javascript
try {
	if (log2consol) console.log("myTimer: opening bus...  Time: " + new Date( _readTime ).toISOString().slice(11, -1) );
	let   _aBus = null;

	// start async error handling: << does not work either
	// this.on('error', function( i2cErr ) { timerErrorHandler(i2cErr) } );
	let ipA;
	let ipB;

	_aBus = i2cModule.openSync(globalChipNode.busNum + 999); // << FORCING an ERROR ... should stop at this line ! 
	if (_aBus) { // << 2. problem:  _aBus is not null, even, if it could not open the bus !
		if (log2consol) console.log("OK. Bus Opened. Now reading A+B banks... Typeof _aBUS:" + typeof(_aBus));
		_processState = 1;
		ipA = _aBus.readByteSync(_addr, 0x12);
		ipB = _aBus.readByteSync(_addr, 0x13);
		if (log2consol) console.log("Read success ipA=" + ipA + "  ipB=" + ipB);
				
		_processState = 3;
		_aBus.closeSync();
...
```

2. problem:  _aBus is not null, even, if it could not open the bus !

PizzaProgram, 2021-09-14:
The log shows:
```
myTimer: opening bus...  Time: 10:04:45.299
OK. Bus Opened. Now reading A+B banks... Typeof _aBUS:object
MPC read error... starting error handler
Reading current state failed.  [Bus=11]  [Addr=39]   [theChip.lastRead=0]
14 Sep 12:04:45 - [error] [mcp23017chip:74d76478.577bec] Error: ENOENT: no such file or directory, open '/dev/i2c-1010'

myTimer: opening bus...  Time: 10:04:45.802
OK. Bus Opened. Now reading A+B banks... Typeof _aBUS:object
MPC read error... starting error handler
Reading current state failed.  [Bus=11]  [Addr=39]   [theChip.lastRead=0]
14 Sep 12:04:45 - [error] [mcp23017chip:74d76478.577bec] Error: ENOENT: no such file or directory, open '/dev/i2c-1010'
...
```

PizzaProgram, 2021-09-16:
Tried to use await with async :
```javascript
const _aBus = await i2cModule.open(globalChipNode.busNum, err1 => { ...
```
But the start result is:
`[warn] [node-red-contrib-mcp23017chip/mcp23017chip] SyntaxError: await is only valid in async function`

fivdi, 2021-09-16:
The [openSync](https://github.com/fivdi/i2c-bus#opensyncbusnumber--options) function is synchronous and not asynchronous. I think what is causing the confusion in this particular case is that `openSync` doesn't actually try to open the non-existant file `'/dev/i2c-1010'`. An attempt is not made to open `'/dev/i2c-1010'` until a function is called that needs to access the file. In the code posted above, the attempt to open `'/dev/i2c-1010'` will be made when `readByteSync` is called.

As mentioned in the documentation for [openSync](https://github.com/fivdi/i2c-bus#opensyncbusnumber--options), it returns a new `Bus` object. It will never return `null`.

fivdi, 2021-09-16:
> Tried to use await with async :
> 
> ```js
> const _aBus = await i2cModule.open(globalChipNode.busNum, err1 => { ...
> ```
> 
> But the node-red-start result is:
> `[warn] [node-red-contrib-mcp23017chip/mcp23017chip] SyntaxError: await is only valid in async function`

To use promises/async/await please call function [openPromisified](https://github.com/fivdi/i2c-bus#openpromisifiedbusnumber--options) as mentioned in the documentation. The `open` function is for asynchronous callbacks based programming.

PizzaProgram, 2021-09-17:
> To use promises/async/await please call function openPromisified

Thanks, I was just about to rework the whole thing to promise-calls.
I will test it in the following days how the error capturing works and report back.

PizzaProgram, 2021-09-17:
Actually I gave an other try to this sync method, and wanted to force-check if the Bus itself is valid.

Found an earlier ticket: https://github.com/fivdi/i2c-bus/issues/69#issuecomment-480518417
... where @toddbu wrote:
A `readByte(0x03, 0x00, cb);` could be used to check, if the Bus is valid or not.
I've also implemented it, but it gives **always the same I/O error**, even for valid working buses too. :-(

# Is there a way to check if the bus itself is valid and working?

(Before trying to poke a device on the bus.)

fivdi, 2021-09-17:
>Is there a way to check if the bus itself is valid and working?

I'm not aware of a generic way to achieve this.

> A readByte(0x03, 0x00, cb); could be used to check, if the Bus is valid or not.

This will only function if there is a device at I2C bus address 0x03 that supports command 0x00.

If your system has a device at a known address, you could attempt to access that device to see if the device is there.

PizzaProgram, 2021-09-23:
> This will only function if there is a device at I2C bus address 0x03 that supports command 0x00.

Thanks! That's what I was afraid of.
Sometimes users may set the wrong bus number, other time the wrong address.
That's why I wanted to test both, step by step.
But it seems it is impossible. The whole I2C standard is lame.

closed 2021-09-23 by PizzaProgram