fivdi/i2c-bus #106: index.d.ts

PizzaProgram, 2021-09-23:
Please include `index.d.ts` in the main directory because it helps a lot to understand definitions about this module.

_It took me 1 year to realize it exists! 
It would have spared me 10+ hours.
It would help others too, who are not using VScode from the beginning, like me._

... or at least mention it at the Readme.md file with a link to it.

Thanks :-)

fivdi, 2021-09-25:
Which `index.d.ts` is being referred to? Is it the one [here](https://github.com/DefinitelyTyped/DefinitelyTyped/tree/master/types/i2c-bus)?

PizzaProgram, 2021-09-25:
YES.

PizzaProgram, 2021-09-25:
Thanks :) Perfect!

closed 2021-09-25 by PizzaProgram

fivdi, 2021-09-25:
Section [TypeScript Type Definitions](https://github.com/fivdi/i2c-bus#typescript-type-definitions) was added to the README to document the availability of the TypeScript type definitions.

> Thanks :) Perfect!

You're welcome.