fivdi/i2c-bus #111: HC-SR04 Ultrasonic Sensor example on i2c-bus

voidbrain, 2022-03-20:
Is it possible to transpose the HC-SR04 Ultrasonic Sensor from pigpio to this plugin? 
I would like to put a number of sensor under a mcp23017 expander.
Please

fivdi, 2022-03-20:
> Is it possible to transpose the HC-SR04 Ultrasonic Sensor from pigpio to this plugin?
> I would like to put a number of sensor under a mcp23017 expander.

I'm afraid not. I2C will not be fast enough and is unsuitable for measuring the width of the short pulses generated by HC-SR04 ultrasonic sensors.

voidbrain, 2022-03-21:
I see. :/ Thank you Sir.

Do you know if there is a way to make it work with an expander? 
A suitable protocol to do that? (SPI protocol should be faster than I2C if i'm not wrong)

fivdi, 2022-03-21:
> Do you know if there is a way to make it work with an expander?

I'm not aware of an I2C or SPI expander that can be used to achieve this. That being said, their may be I2C or SPI devices that are suitable for measuring pulse widths. I haven't searched for such a device.

> A suitable protocol to do that? (SPI protocol should be faster than I2C if i'm not wrong)

SPI, like I2C, is a communication protocol. It's focus is transferring information rather than measuring the width of pulses.

closed 2022-03-21 by fivdi