fivdi/i2c-bus #112: writeI2cBlockSync returns a PromisifiedBus rather than the suggested number of bytes written

reesesm2000, 2022-08-07:
Calling writeI2cBlockSync correctly writes data to the device but rather than returning what I assume should be the length written (based on VS intellisense saying the function returns a number), it returns a PromisifiedBus object. What should be returned by this function?

fivdi, 2022-08-07:
The documentation for `writeI2cBlockSync` can be found [here](https://github.com/fivdi/i2c-bus#buswritei2cblocksyncaddr-cmd-length-buffer). The documentation doesn't mention a return value so it should be assumed there is none. `writeI2cBlockSync` doesn't have a documented return value as the C function that ends up being called by `writeI2cBlockSync` doesn't document it's return value either. If `writeI2cBlockSync` fails, an error will be thrown.

In reality, as can be seen [here](https://github.com/fivdi/i2c-bus/blob/v5.2.2/i2c-bus.js#L525), `writeI2cBlockSync` returns the bus object that it was called on. Note that this is undocumented and shouldn't be relied on. You mention above that when you call `writeI2cBlockSync`, a `PromisifiedBus` is returned which implies that your program is using a `PromisifiedBus`. The documentation for `PromisifiedBus` can be found [here](https://github.com/fivdi/i2c-bus#class-promisifiedbus) and as can be seen, it is not documented that it supports method `writeI2cBlockSync`. 

I hope this answers your question.

reesesm2000, 2022-08-07:
Thanks for the super quick response!
Thanks for the explanation. I had just assumed (which I hate doing lol) that since VS intellisense was showing a number being returned, it would most likely be the length. Below is a screenshot of what it shows.
I will close this now.

[image]

closed 2022-08-07 by reesesm2000

fivdi, 2022-08-07:
I see you are using TypeScript. The type definitions for i2c-bus don't appear to be 100% correct, for example, `writeI2cBlockSync` is defined like [this](https://github.com/DefinitelyTyped/DefinitelyTyped/blob/master/types/i2c-bus/index.d.ts#L427-L443) and specifies that the number of bytes written is returned as a number which is incorrect.

reesesm2000, 2022-08-07:
Yeah, forgot to mention that. So `writeI2cBlockSync` should probably return void like `writeI2cBlock` does.

fivdi, 2022-08-07:
Yes, that would be correct.