fivdi/i2c-bus #116: Failing to install on Raspberry Pi Zero 2 W

gunnarsson, 2023-08-06:
I'm having a hard time installing i2c-bus on my Raspberry Pi Zero 2 W. I'm running Node 20.4.0 and npm 9.8.1. The docs clearly state that Node versions 10, 12, 14, 15 and 16 are supported, but I can install the module locally on my Mac with Node 20 without any issues. Is there anything I can do to work around this problem myself?

Here's what I believe is the relevant part of the npm log:
```
81 error gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
81 error In file included from ../../nan/nan.h:178,
81 error                  from ../src/i2c.cc:2:
81 error ../../nan/nan_callbacks.h:55:23: error: ‘AccessorSignature’ is not a member of ‘v8’
81 error    55 | typedef v8::Local<v8::AccessorSignature> Sig;
81 error       |                       ^~~~~~~~~~~~~~~~~
81 error ../../nan/nan_callbacks.h:55:40: error: template argument 1 is invalid
81 error    55 | typedef v8::Local<v8::AccessorSignature> Sig;
81 error       |                                        ^
81 error In file included from ../src/i2c.cc:2:
81 error ../../nan/nan.h: In function ‘void Nan::SetAccessor(v8::Local<v8::ObjectTemplate>, v8::Local<v8::String>, Nan::GetterCallback, Nan::SetterCallback, v8::Local<v8::Value>, v8::AccessControl, v8::PropertyAttribute, Nan::imp::Sig)’:
81 error ../../nan/nan.h:2551:16: error: no matching function for call to ‘v8::ObjectTemplate::SetAccessor(v8::Local<v8::String>&, void (*&)(v8::Local<v8::Name>, const v8::PropertyCallbackInfo<v8::Value>&), void (*&)(v8::Local<v8::Name>, v8::Local<v8::V>
81 error  2551 |     , signature);
81 error       |                ^
81 error In file included from /home/martin/.cache/node-gyp/20.4.0/include/node/v8-function.h:15,
81 error                  from /home/martin/.cache/node-gyp/20.4.0/include/node/v8.h:33,
81 error                  from /home/martin/.cache/node-gyp/20.4.0/include/node/node.h:73,
81 error                  from ../src/i2c.cc:1:
81 error /home/martin/.cache/node-gyp/20.4.0/include/node/v8-template.h:809:8: note: candidate: ‘void v8::ObjectTemplate::SetAccessor(v8::Local<v8::String>, v8::AccessorGetterCallback, v8::AccessorSetterCallback, v8::Local<v8::Value>, v8::AccessControl,>
81 error   809 |   void SetAccessor(
81 error       |        ^~~~~~~~~~~
81 error /home/martin/.cache/node-gyp/20.4.0/include/node/v8-template.h:814:22: note:   no known conversion for argument 7 from ‘Nan::imp::Sig’ {aka ‘int’} to ‘v8::SideEffectType’
81 error   814 |       SideEffectType getter_side_effect_type = SideEffectType::kHasSideEffect,
81 error       |       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
81 error /home/martin/.cache/node-gyp/20.4.0/include/node/v8-template.h:816:8: note: candidate: ‘void v8::ObjectTemplate::SetAccessor(v8::Local<v8::Name>, v8::AccessorNameGetterCallback, v8::AccessorNameSetterCallback, v8::Local<v8::Value>, v8::AccessCo>
81 error   816 |   void SetAccessor(
81 error       |        ^~~~~~~~~~~
81 error /home/martin/.cache/node-gyp/20.4.0/include/node/v8-template.h:821:22: note:   no known conversion for argument 7 from ‘Nan::imp::Sig’ {aka ‘int’} to ‘v8::SideEffectType’
81 error   821 |       SideEffectType getter_side_effect_type = SideEffectType::kHasSideEffect,
81 error       |       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
81 error make: *** [i2c.target.mk:115: Release/obj.target/i2c/src/i2c.o] Error 1
81 error gyp ERR! build error
81 error gyp ERR! stack Error: `make` failed with exit code: 2
81 error gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:203:23)
81 error gyp ERR! stack     at ChildProcess.emit (node:events:512:28)
81 error gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_process:293:12)
81 error gyp ERR! System Linux 6.1.21-v7+
81 error gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
81 error gyp ERR! cwd /home/martin/subby/node_modules/i2c-bus
81 error gyp ERR! node -v v20.4.0
81 error gyp ERR! node-gyp -v v9.4.0
81 error gyp ERR! not ok
```

Any help would be much appreciated!
Thanks,
Martin

fivdi, 2023-08-06:
I don't have a Raspberry Pi Zero 2 W and I'm afraid I can't reproduce the error on a Raspberry Pi 4.

If I run the following commands on a Raspberry Pi 4:

```
node -v
npm -v
uname -a
npm install i2c-bus
npm list --all
node node_modules/i2c-bus/example/i2cdetect
node node_modules/i2c-bus/example/i2cfuncs.js 
```

Everything functions as expected and this is what is see:

```
pi@raspberrypi:~/test $ node -v
v20.4.0
pi@raspberrypi:~/test $ npm -v
9.7.2
pi@raspberrypi:~/test $ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
pi@raspberrypi:~/test $ npm install i2c-bus

added 4 packages in 12s
pi@raspberrypi:~/test $ npm list --all
test@ /home/pi/test
└─┬ i2c-bus@5.2.2
  ├─┬ bindings@1.5.0
  │ └── file-uri-to-path@1.0.0
  └── nan@2.17.0

pi@raspberrypi:~/test $ node node_modules/i2c-bus/example/i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --                        
pi@raspberrypi:~/test $ node node_modules/i2c-bus/example/i2cfuncs.js 
Functionalities implemented by /dev/i2c-1:
I2C                              yes
SMBus Quick Command              yes
SMBus Send Byte                  yes
SMBus Receive Byte               yes
SMBus Write Byte                 yes
SMBus Read Byte                  yes
SMBus Write Word                 yes
SMBus Read Word                  yes
SMBus Process Call               yes
SMBus Block Write                yes
SMBus Block Read                 no
SMBus Block Process Call         no
SMBus PEC                        yes
I2C Block Write                  yes
I2C Block Read                   yes
pi@raspberrypi:~/test $ 
```

Are you perhaps installing an old version of i2c-bus?
Exactly what command(s) are you using to install i2c-bus?
If you run the commands shown above, exactly as shown, from an empty newly created directory, what is the complete output?

gunnarsson, 2023-08-06:
Thanks for looking into this so quickly! I tried what you suggested (plus an `npm init` in the directory) and everything works perfectly. The directory where it doesn't is an existing Node project, so I assume one of the other modules has incompatible dependencies.

After a quick look through the package tree, I think the culprit might be:
```
├─┬ pigpio@3.3.1
│ ├─┬ bindings@1.5.0
│ │ └── file-uri-to-path@1.0.0
│ └── nan@2.15.0
```

... as my test directory has `nan@2.17.0`. I'll fiddle around a little with it, thanks for your time!

gunnarsson, 2023-08-06:
For anybody else running into this issue, manually installing the correct version of nan and then re-running the i2c-bus installation seems to have done the trick. I haven't had time to verify the installation yet, but at least it went through without any error messages.

fivdi, 2023-08-06:
@gunnarsson Thank you for this additional information and for opening the issue initially. I'll reopen the issue as there is a problem with the i2c-bus dependencies. As can be seen [here](https://github.com/fivdi/i2c-bus/blob/v5.2.2/package.json#L27) i2c-bus@5.2.2 depends on nan version "^2.14.2". This dependency needs to be updated to "^2.17.0".

gunnarsson, 2023-08-06:
All right, that explains why it failed in the first place given these circumstances. Thanks again for looking into this!

fivdi, 2023-08-06:
> All right, that explains why it failed in the first place given these circumstances. Thanks again for looking into this!

Yes, that explains it.

fivdi, 2023-08-06:
The problem should be resolved with i2c-bus@5.2.3 which has just been published on npm.

@gunnarsson Once again, thank you for reporting this issue.

closed 2023-08-06 by fivdi

gunnarsson, 2023-08-07:
Fantastic! I re-started with the new version, and I can confirm that the installation went smoothly.