fivdi/i2c-bus #118: Asistance with reading a register

john-, 2023-11-10:
I am porting an application from perl to node and bringing over the I2C part.  In perl, one of the I2C libraries has code like this:
```
async method read_shunt_voltage ()
{
   my $vraw = unpack "s>", await $self->read_reg( REG_VSHUNT, 1 );
 
   # Each $vraw graduation is 10uV
   return $vraw * 10;
}
```
With i2c-bus my code stands like this:
```
  this.i2cCheck = setInterval(() => {
    const INA219_ADDR = 0x40
    const VSHUNT_REG = 0x01
    openPromisified(0).
    then(i2c1 => i2c1.readWord(INA219_ADDR, VSHUNT_REG).
      then((rawData: any) => console.log(rawData)).
      //then((rawData: any) => console.log(toCelsius(rawData))).
      then(_ => i2c1.close())
    ).catch(console.log);

  }, 5000)
```
Although this prints values there are two things I need to confirm/address:
1.  I am pretty sure `readWord` does the same thing as what `read_reg` is doing.
2. The perl code uses `unpack "s>" to unpack a big-endian signed short (16bit).  What is the equivalent thing in javascript?  I  am having a heard time getting by head around the bit manipulation in the i2c-bus examples and not even sure if `readWord` is the correct starting point.

Any help would be appreciated.

fivdi, 2023-11-11:
Method `readWord` assumes the 16 bit values read are little-endian because this is how I2C/SMBUS expects it to be. On most devices that I have used 16 bit values are actually little-endian and `readWord` functions as expected. With some devices 16 bit values are big-endian in which case the two bytes in the 16 bit word need to be swapped with something like this:
```js
  rawData = (rawData >> 8) + ((rawData & 0xff) << 8);
```

fivdi, 2023-11-11:
For a related discussion see #82

john-, 2023-11-11:
Thanks...that fixed it for me.  Also, I will take a deeper look at that thread as it makes a bit more sense to me now.

closed 2023-11-11 by john-