fivdi/i2c-bus #119: Approach for device abstraction layer

john-, 2023-11-11:
In the perl world there is a namespace for device specific classes that use I2C.  For example, the [INA219](https://metacpan.org/pod/Device::Chip::INA219).  These wrap up methods to change configuration and get device specific values.  They abstract the underlying I2C library.  I don't see anything like this for i2c-bus.

Is there something people are doing in their projects?  Any thoughts on the approach I should take?

john-, 2023-11-11:
My first attempt is [here](https://gitlab.com/john---/cart_dashboard/-/blob/main/BackEnd/src/models/telemetry/INA219.ts?ref_type=heads).

Instantiated like this:
```
const ina219 = new INA219( {bus: 0, addr: 0x40})
```
Get a value:
```
const milliVolts = await ina219.readShuntVoltage()
```

I am not yet comfortable with promise based javascript but it at least works.

Thanks for much for `i2c-bus`.  It saved me a lot of time if I could even do it at all!

fivdi, 2023-11-12:
> In the perl world there is a namespace for device specific classes that use I2C. For example, the [INA219](https://metacpan.org/pod/Device::Chip::INA219). These wrap up methods to change [...]

JavaScript lacks namespaces which bascially provide a scope to prevent collisions between identifiers/names. In the Node.js world there is the [npm registry](https://www.npmjs.com/). A search of the npm registry indicates that there appear to be several packages offering support for the INA219. There are also packages for other I2C devices available in the npm registry.

john-, 2023-11-12:
Good to know, thanks.   I have a few other devices I need to implement and will take a look at the other options before I get deeper into it.

closed 2023-11-12 by john-

fivdi, 2023-11-12:
> My first attempt is [here](https://gitlab.com/john---/cart_dashboard/-/blob/main/BackEnd/src/models/telemetry/INA219.ts?ref_type=heads).

One thing that sticks out is that the implementation calls [`openPromisified()`](https://github.com/fivdi/i2c-bus#openpromisifiedbusnumber--options) which yields a `PromisifiedBus` object, however, it never calls [`promisifiedBus.close()`](https://github.com/fivdi/i2c-bus#promisifiedbusclose) to free the resources allocated by the operating system while the I2C bus is being used by the application.

> Thanks for much for `i2c-bus`. It saved me a lot of time if I could even do it at all!

Thank you :)

john-, 2023-11-12:
I appreciate the feedback on the code!