fivdi/i2c-bus #120: Error: EIO, Input/output error

nbgooch, 2023-11-20:
Can someone please help me with this issue. 

My device is visible over i2c-1
```
nbgooch@raspberrypi:~/code/diskdoctor/packages/backend $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- 1c -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- 42 -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- -- 
70: -- -- -- -- -- -- -- 77
```
but if I try 
```
      this.wire = i2c.openSync(1);
      console.log(`Scanned: ${this.wire.scanSync()}`);
      console.log(`Device: ${this.wire.deviceIdSync(0x1C)}`);
```
I get 
```
Scanned: 28,66,106,119
Failed to create accelerometer Error: EIO, Input/output error
```
I tried a few other libs and I am worried it is a permissions error, but would love some help.

fivdi, 2023-11-20:
The issue you are seeing here may be the same as or similar to #76.

`deviceId` will only work if the I2C device supports it. For further information please take a look at the datasheet for the I2C device in question. Devices are not required to support `deviceId`. Does your device support `deviceId`?

nbgooch, 2023-11-22:
Yes agreed found that post after creating this issue. Seems like `deviceId` is not supported on `LSM6DSL`, but not sure how to confirm. Thanks for the quick response.

closed 2023-11-22 by nbgooch