fivdi/i2c-bus #122: Node 22 support

baybal, 2024-09-04:
Node 22 removed a number of APIs used by i2c-bus

fivdi, 2024-09-06:
Works fine for me.

Please provide the complete details required to reproduce the issue.

Links2004, 2024-09-06:
same here, does not build on a raspberry pi.

```
npm error code 1
npm error path /root/test/node_modules/i2c-bus
npm error command failed
npm error command sh -c node-gyp rebuild
npm error make: Entering directory '/root/test/node_modules/i2c-bus/build'
npm error   CXX(target) Release/obj.target/i2c/src/i2c.o
npm error make: Leaving directory '/root/test/node_modules/i2c-bus/build'
npm error gyp info it worked if it ends with ok
npm error gyp info using node-gyp@10.1.0
npm error gyp info using node@22.8.0 | linux | arm
npm error gyp info find Python using Python version 3.9.2 found at "/usr/bin/python3"
npm error gyp info spawn /usr/bin/python3
npm error gyp info spawn args [
npm error gyp info spawn args '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm error gyp info spawn args 'binding.gyp',
npm error gyp info spawn args '-f',
npm error gyp info spawn args 'make',
npm error gyp info spawn args '-I',
npm error gyp info spawn args '/root/test/node_modules/i2c-bus/build/config.gypi',
npm error gyp info spawn args '-I',
npm error gyp info spawn args '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm error gyp info spawn args '-I',
npm error gyp info spawn args '/root/.cache/node-gyp/22.8.0/include/node/common.gypi',
npm error gyp info spawn args '-Dlibrary=shared_library',
npm error gyp info spawn args '-Dvisibility=default',
npm error gyp info spawn args '-Dnode_root_dir=/root/.cache/node-gyp/22.8.0',
npm error gyp info spawn args '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
npm error gyp info spawn args '-Dnode_lib_file=/root/.cache/node-gyp/22.8.0/<(target_arch)/node.lib',
npm error gyp info spawn args '-Dmodule_root_dir=/root/test/node_modules/i2c-bus',
npm error gyp info spawn args '-Dnode_engine=v8',
npm error gyp info spawn args '--depth=.',
npm error gyp info spawn args '--no-parallel',
npm error gyp info spawn args '--generator-output',
npm error gyp info spawn args 'build',
npm error gyp info spawn args '-Goutput_dir=.'
npm error gyp info spawn args ]
npm error gyp info spawn make
npm error gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm error In file included from ../src/i2c.cc:2:
npm error ../../nan/nan.h: In function ‘void Nan::SetAccessor(v8::Local<v8::ObjectTemplate>, v8::Local<v8::String>, Nan::GetterCallback, Nan::SetterCallback, v8::Local<v8::Value>, v8::AccessControl, v8::PropertyAttribute, Nan::imp::Sig)’:
npm error ../../nan/nan.h:2558:3: error: no matching function for call to ‘v8::ObjectTemplate::SetAccessor(v8::Local<v8::String>&, void (*&)(v8::Local<v8::Name>, const v8::PropertyCallbackInfo<v8::Value>&), void (*&)(v8::Local<v8::Name>, v8::Local<v8::Value>, const v8::PropertyCallbackInfo<void>&), v8::Local<v8::Object>&, v8::AccessControl&, v8::PropertyAttribute&)’
npm error  2558 |   );
npm error       |   ^
npm error In file included from /root/.cache/node-gyp/22.8.0/include/node/v8-function.h:15,
npm error                  from /root/.cache/node-gyp/22.8.0/include/node/v8.h:33,
npm error                  from /root/.cache/node-gyp/22.8.0/include/node/node.h:73,
npm error                  from ../src/i2c.cc:1:
npm error /root/.cache/node-gyp/22.8.0/include/node/v8-template.h:1049:8: note: candidate: ‘void v8::ObjectTemplate::SetAccessor(v8::Local<v8::String>, v8::AccessorGetterCallback, v8::AccessorSetterCallback, v8::Local<v8::Value>, v8::PropertyAttribute, v8::SideEffectType, v8::SideEffectType)’
npm error  1049 |   void SetAccessor(
npm error       |        ^~~~~~~~~~~
npm error /root/.cache/node-gyp/22.8.0/include/node/v8-template.h:1052:61: note:   no known conversion for argument 5 from ‘v8::AccessControl’ to ‘v8::PropertyAttribute’
npm error  1052 |       Local<Value> data = Local<Value>(), PropertyAttribute attribute = None,
npm error       |                                           ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
npm error /root/.cache/node-gyp/22.8.0/include/node/v8-template.h:1055:8: note: candidate: ‘void v8::ObjectTemplate::SetAccessor(v8::Local<v8::Name>, v8::AccessorNameGetterCallback, v8::AccessorNameSetterCallback, v8::Local<v8::Value>, v8::PropertyAttribute, v8::SideEffectType, v8::SideEffectType)’
npm error  1055 |   void SetAccessor(
npm error       |        ^~~~~~~~~~~
npm error /root/.cache/node-gyp/22.8.0/include/node/v8-template.h:1058:61: note:   no known conversion for argument 5 from ‘v8::AccessControl’ to ‘v8::PropertyAttribute’
npm error  1058 |       Local<Value> data = Local<Value>(), PropertyAttribute attribute = None,
npm error       |                                           ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
npm error In file included from ../src/i2c.cc:2:
npm error ../../nan/nan.h: In function ‘void Nan::SetAccessor(v8::Local<v8::ObjectTemplate>, v8::Local<v8::String>, Nan::GetterCallback, Nan::SetterCallback, v8::Local<v8::Value>, v8::AccessControl, v8::PropertyAttribute)’:
npm error ../../nan/nan.h:2601:3: error: no matching function for call to ‘v8::ObjectTemplate::SetAccessor(v8::Local<v8::String>&, void (*&)(v8::Local<v8::Name>, const v8::PropertyCallbackInfo<v8::Value>&), void (*&)(v8::Local<v8::Name>, v8::Local<v8::Value>, const v8::PropertyCallbackInfo<void>&), v8::Local<v8::Object>&, v8::AccessControl&, v8::PropertyAttribute&)’
npm error  2601 |   );
npm error       |   ^
npm error In file included from /root/.cache/node-gyp/22.8.0/include/node/v8-function.h:15,
npm error                  from /root/.cache/node-gyp/22.8.0/include/node/v8.h:33,
[... cut ...]
npm error gyp ERR! stack Error: `make` failed with exit code: 2
npm error gyp ERR! stack at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:209:23)
npm error gyp ERR! System Linux 6.1.21-v7+
npm error gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm error gyp ERR! cwd /root/test/node_modules/i2c-bus
npm error gyp ERR! node -v v22.8.0
npm error gyp ERR! node-gyp -v v10.1.0
npm error gyp ERR! not ok
```

fivdi, 2024-09-06:
Here is what I see when i2c-bus is installed and the `i2cdetect` example program is run on a Raspberry Pi using Node.js v22.8.0:
```
pi@raspberrypi:~ $ node --version
v22.8.0
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
pi@raspberrypi:~ $ mkdir test
pi@raspberrypi:~ $ cd test
pi@raspberrypi:~/test $ npm install i2c-bus

added 4 packages in 4s
pi@raspberrypi:~/test $ node node_modules/i2c-bus/example/i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- 18 -- -- -- -- -- 1e --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- 77                        
pi@raspberrypi:~/test $ 
```
As can be seen, everything works as expected.

@Links2004 I see that your are attempting to install i2c-bus in `/root/test`. The issues you are having may be related to privileges rather than i2c-bus.

Links2004, 2024-09-07:
thanks for testing, got it fixed by deleting the node-gyp cache (`$HOME/.cache/node-gyp`) now its working.
tests where done as root user, so no access problems.

fivdi, 2025-04-13:
I'll go ahead and close this as i2c-bus function successfully with Node 22 and Node 23.