fivdi/i2c-bus #126: N-API version of i2c-bus

Teakzieas, 2024-12-21:
I have been utilizing your i2c-bus package for I2C communication in Node.js and greatly appreciate your contributions to the community.

I am interested in exploring I2C packages that leverage N-API for improved stability and compatibility across Node.js versions. Could you recommend any I2C packages that are built with N-API? Additionally, are there any plans to migrate i2c-bus to N-API in the future?

Thank you for your time and assistance.

Best regards,

fivdi, 2025-04-13:
Let me start off by apoligizing for late reply.

> I have been utilizing your i2c-bus package for I2C communication in Node.js and greatly appreciate your contributions to the community.

Thank you very much.

> I am interested in exploring I2C packages that leverage N-API for improved stability and compatibility across Node.js versions. Could you recommend any I2C packages that are built with N-API?

I am not aware of any I2C packages that are built with N-API.

> Additionally, are there any plans to migrate i2c-bus to N-API in the future?

I don't have any plans to do this.

Teakzieas, 2025-04-18:
Thank you anyways :), it's a nice repo

closed 2025-04-18 by Teakzieas