fivdi/i2c-bus #128: Installation problems on Luckfox Pico Max

DeroZaza, 2025-03-02:
`pico@luckfox:~/pico$ npm install i2c-bus
npm error code 1
npm error path /home/pico/pico/node_modules/i2c-bus
npm error command failed
npm error command sh -c node-gyp rebuild
npm error make: Entering directory '/home/pico/pico/node_modules/i2c-bus/build'
npm error   CXX(target) Release/obj.target/i2c/src/i2c.o
npm error   SOLINK_MODULE(target) Release/obj.target/i2c.node
npm error make: Leaving directory '/home/pico/pico/node_modules/i2c-bus/build'
npm error gyp info it worked if it ends with ok
npm error gyp info using node-gyp@10.1.0
npm error gyp info using node@18.20.6 | linux | arm
npm error gyp info find Python using Python version 3.10.12 found at "/usr/bin/python3"
npm error gyp info spawn /usr/bin/python3
npm error gyp info spawn args [
npm error gyp info spawn args '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm error gyp info spawn args 'binding.gyp',
npm error gyp info spawn args '-f',
npm error gyp info spawn args 'make',
npm error gyp info spawn args '-I',
npm error gyp info spawn args '/home/pico/pico/node_modules/i2c-bus/build/config.gypi',
npm error gyp info spawn args '-I',
npm error gyp info spawn args '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm error gyp info spawn args '-I',
npm error gyp info spawn args '/home/pico/.cache/node-gyp/18.20.6/include/node/common.gypi',
npm error gyp info spawn args '-Dlibrary=shared_library',
npm error gyp info spawn args '-Dvisibility=default',
npm error gyp info spawn args '-Dnode_root_dir=/home/pico/.cache/node-gyp/18.20.6',
npm error gyp info spawn args '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
npm error gyp info spawn args '-Dnode_lib_file=/home/pico/.cache/node-gyp/18.20.6/<(target_arch)/node.lib',
npm error gyp info spawn args '-Dmodule_root_dir=/home/pico/pico/node_modules/i2c-bus',
npm error gyp info spawn args '-Dnode_engine=v8',
npm error gyp info spawn args '--depth=.',
npm error gyp info spawn args '--no-parallel',
npm error gyp info spawn args '--generator-output',
npm error gyp info spawn args 'build',
npm error gyp info spawn args '-Goutput_dir=.'
npm error gyp info spawn args ]
npm error gyp info spawn make
npm error gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm error /usr/bin/ld: cannot find -lgcc: No such file or directory
npm error collect2: error: ld returned 1 exit status
npm error make: *** [i2c.target.mk:141: Release/obj.target/i2c.node] Error 1
npm error gyp ERR! build error
npm error gyp ERR! stack Error: `make` failed with exit code: 2
npm error gyp ERR! stack at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:209:23)
npm error gyp ERR! System Linux 5.10.160
npm error gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm error gyp ERR! cwd /home/pico/pico/node_modules/i2c-bus
npm error gyp ERR! node -v v18.20.6
npm error gyp ERR! node-gyp -v v10.1.0
npm error gyp ERR! not ok
npm error A complete log of this run can be found in: /home/pico/.npm/_logs/2025-03-02T16_07_55_560Z-debug-0.log
`

What I already did:

sudo apt install build-essential libgcc-11-dev g++ gcc make

DeroZaza, 2025-03-02:
I got it to work by reinstalling the librarys and g++ compiler:

$ sudo apt-get remove --purge libgcc-11-dev libgcc-10-dev libgcc-9-dev
$ sudo ldconfig
$ ldconfig -p | grep libgcc
$ sudo apt-get update
$ sudo apt-get install g++

fivdi, 2025-04-11:
Good to hear that you got it to work :)

closed 2025-04-11 by fivdi