fivdi/i2c-bus #129: [BUG] Build fails due to NAN incompatibility on latest node

JulianWowra, 2025-04-11:
## ❌ Build fails on Node.js `v23` due to NAN incompatibility

### Description

Attempting to install dependencies using `pnpm install --frozen-lockfile` fails when building `i2c-bus@5.2.3`, due to compatibility issues between `nan` and Node.js v23.11.0. The error seems related to updated V8 APIs and deprecated callback types in `v8-template.h`.

---

### Environment

- **OS:** Linux (CI runner)
- **Node.js:** `v23.11.0`
- **Package manager:** pnpm `10.8.0`
- **`i2c-bus` version:** `5.2.3`
- **`nan` version:** `2.20.0`

---

### Error Log (excerpt)

```
error: invalid conversion from ‘Nan::imp::NativeIndexGetter’ to ‘v8::IndexedPropertyGetterCallbackV2’ [-fpermissive]
...
error: no matching function for call to ‘v8::ScriptOrigin::ScriptOrigin(v8::Isolate*, v8::Local<v8::Value>&)’
```

[Full log.](https://github.com/JulianWowra/node-anemometer/actions/runs/14403993295/job/40395923111)

---

### Steps to Reproduce

1. Use Node.js `v23.11.0`
2. Run `pnpm install` with `i2c-bus@5.2.3` in dependencies
3. Observe build failure during `node-gyp rebuild`

---

### Expected Behavior

Build should complete successfully or an updated release of `i2c-bus` should provide compatibility with Node.js `v23.x.x`

---

### Suggested Fix

It looks like `nan` is not fully compatible with the changes introduced in Node.js `v23`. Updating `nan` (if supported) or refactoring the native bindings in `i2c-bus` to not rely on deprecated V8 APIs might resolve the issue.

---

### Related

Maybe: #122

fivdi, 2025-04-11:
What happens if you install i2c-bus the way I installed it above?

fivdi, 2025-04-11:
You're going to have to figure out why nan@2.20.0 is being used on your system rather than nan@2.22.2.

JulianWowra, 2025-04-11:
> You're going to have to figure out why nan@2.20.0 is being used on your system rather than nan@2.22.2.

How can I find out? Even the Github Pipeline, where this project was installed, comes to the same result. Maybe you try to reproduce the problem my way again?

> What happens if you install i2c-bus the way I installed it above?

```bash
pi@raspberrypi:~/test $ nvm exec v23 npm install i2c-bus
Running node v23.11.0 (npm v10.9.2)

added 4 packages in 32s
```

Successful with npm. A problem with pnpm?

JulianWowra, 2025-04-11:
I did some tests and came to this interesting conclusion:

| **Node.js version** 	| **Command**                 	| **Result** 	|
|---------------------	|-----------------------------	|------------	|
| `v18.16.0`          	| `nvm exec v18 pnpm rebuild` 	|  Success ✅ 	|
| `v20.19.0`          	| `nvm exec v20 pnpm rebuild` 	|  Success ✅ 	|
| `v22.14.0` - *LTS*  	| `nvm exec v22 pnpm rebuild` 	|  Success ✅ 	|
| `v23.11.0`          	| `nvm exec v23 pnpm rebuild` 	|   Failed ❌ 	|

JulianWowra, 2025-04-12:
Works on windows with Node `v23.11.0` and nan `2.20.0`

```powershell
PS E:\test> nvm use v23
v23.11.0
PS E:\test> corepack enable
PS E:\test> pnpm rebuild  
node_modules/.pnpm/i2c-bus@5.2.3/node_modules/i2c-bus: Running install script, done in 22.7s
PS E:\test> pnpm why nan
dependencies:
i2c-bus 5.2.3
└── nan 2.20.0
```

I am confused... 👀

fivdi, 2025-04-13:
> Works on windows with Node `v23.11.0` and nan `2.20.0`
> 
> PS E:\test> nvm use v23
> v23.11.0
> PS E:\test> corepack enable
> PS E:\test> pnpm rebuild  
> node_modules/.pnpm/i2c-bus@5.2.3/node_modules/i2c-bus: Running install script, done in 22.7s
> PS E:\test> pnpm why nan
> dependencies:
> i2c-bus 5.2.3
> └── nan 2.20.0
> I am confused... 👀

This is to be expected as the binding.gyp file is written is such a way that the code in i2c.cc is only compiled on Linux systems. See [here](https://github.com/fivdi/i2c-bus/blob/d6a42496d0b75a93dc5f53d27a405658eaae261c/binding.gyp#L8-L11). Because i2c.cc is not compiled on Windows, you will not see the compile errors you are seeing on Linux systems. The idea here is to allow people to develop their Node.js applications using i2c-bus on Macs or Windows PCs without having unnecessary compile problems.

JulianWowra, 2025-04-13:
The problem is solved.

`nan@2.20.0` which was installed again and again due to my `pnpm-lock.yml` is only compatible up to Node.js version 20. Partly also with Node.js 22-LTS. Therefore I could rebuild with pnpm with lower Node.js versions. With npm it worked without any problems, of course, because this does not access the `pnpm-lock.yml`.

After deleting the `pnpm-lock.yml` and the `node_modules/` directory, the current `nan@2.22.2` was installed and the build was done.

Thank you for your help!

With kind regards
🇩🇪 Julian

closed 2025-04-13 by JulianWowra

fivdi, 2025-04-13:
Good to hear that it now works.